awaredu/taller-arduino-lora-scrum #11: Problema al subir el código al arduino y no estoy seguro de si me falta algún paso de THN y Cayenne

MikiSD, 2017-11-20:
Buenas tardes, he configurado creo que todo bien en el Arduino, hice en the things network la aplicación con y obtuve el DeviceEUI la Application EUI, y la app key. añadí el device, hice la integración con cayenne y el payload de cayenne también.  Voy a cargar el código en el Arduino modificando los parámetros de Device EUI, etc poniendo los que tengo. Tengo instaladas las librerías del sensor, del arduino y del cayenne en teoría. Al intentar subir al arduino el código me dice "exit status 1 error compilando para la tarjeta Seeedunio lorawan".  No sé a que puede ser debido a vote pronto.  Me creé la cuenta en mydevcies, en Cayenne y creo que añadí el arduino también.  Algo me debe faltar imagino.  De todas formas tengo que solucionar primero el error de subir el código al arduino.  Si alguien sabe el porque no me sube el código, gracias.  Si ponen el tutorial en vídeo para poder visualizarlo y ver en que fallé gracias. A ver si lo puedo solucionar para intentar ver si estoy a rango del Gateway del Cabildo.

nalbet, 2017-11-20:
Hola! Nos puedes copiar más sobre el error para tener más pistas? (si no te sale más, puedes ir a Archivo->Preferencias y que Muestre detalles durante la compilación).
Por otro lado, dices que tienes las librerías instaladas, en teoría... Las puedes ver como instaladas si vas al gestor de librerías? (en sketch -> añadir librería -> gestionar librerías)

MikiSD, 2017-11-21:
Buenas noches, ya resolví el problema alguna de las librerías estaba dando conflicto probablemente la de cayenne, las volvía a instalar y ya conseguí programar el arduino con los datos y en el monitor de arduido da los datos y rula, el problema ahora es que en the things network ni en cayenne me muestra los datos.  No sé si es que me flata algo por esa parte o que el gateway del cabildo no lo pilla, incluso fuí a la zona del  lidll y no rula.

marcello-lodi, 2017-11-22:
Hola, yo solo te puedo decir que de San Antonio (Breña baja) consigo conectarme sin ningún problema al gateway del cabildo. Pero aunque la distancia es mayor, también es cierto que tengo casi una vision directa.

MikiSD, 2017-11-22:
Pues algo me falta entonces, tendré que seguir intentando a ver. Graciass