awaredu/taller-arduino-lora-scrum #28: Alguién me puede ayudar con el sensor DS18B20 (o no es el DS18B20?)

MikiSD, 2017-12-04:
No consigo que rule, este es el código si alguien puede echarle un vistazo (Entiendo que el Sensor que me dejó Angel es el DS18B20):

#include <OneWire.h>
#include <DallasTemperature.h>

// Data wire is plugged into pin 2 on the Arduino

#define ONE_WIRE_BUS 2

// Setup a oneWire instance to communicate with any OneWire devices (not just Maxim/Dallas temperature ICs)

OneWire oneWire(ONE_WIRE_BUS);

// Pass our oneWire reference to Dallas Temperature.

DallasTemperature sensors(&oneWire);
 
void setup(void)

{

// start serial port

Serial.begin(9600);

Serial.println("Dallas Temperature IC Control Library Demo");
 
// Start up the library

sensors.begin(); // IC Default 9 bit. If you have troubles consider upping it 12. Ups the delay giving the IC more time to process the temperature measurement

}

void loop(void)

{

// call sensors.requestTemperatures() to issue a global temperature

// request to all devices on the bus

Serial.print("Requesting temperatures...");

sensors.requestTemperatures(); // Send the command to get temperatures
Serial.println("DONE");
 
Serial.print("Temperature for Device 1 is: ");
Serial.print(sensors.getTempCByIndex(0)); // Why "byIndex"? You can have more than one IC on the same bus. 0 refers to the first IC on the wire

}

manglez, 2017-12-05:
Hola Michael,

El sensor es este?: 
https://www.seeedstudio.com/category/Soil-Moisture-%26-Temperature-Sensor-p-1356.html

Si es así no es el DS18B20 sino el SHT10 de Sensirion.

Creo que las librerias de arduino se llaman SHT1x o algo parecido. Con buscar SHT seguro que te aparecen.

MikiSD, 2017-12-05:
Ese parece que lo tengo controlado, el que no controlo es este https://www.makerlab-electronics.com/product/waterproof-temperature-sensor-ds18b20/ que se supone sería más sencilllo

MikiSD, 2017-12-06:
Resuelto. Le faltaba en los Serial.print poner SerialUSB.print, Gracias a la ayuda de Manuel se arregló el problema.

closed 2017-12-06 by MikiSD